ros2/system_tests #7: OpenSplice Logging

wjwwood, 2015-06-08:
Currently on OS X several of these tests fail:

http://54.183.26.131:8080/job/ros2_batch_ci_osx/149/testReport/

I've been trying to track down the failures with OpenSplice, but all of the opensplice processes simultaneously dump their logs to the same file:

http://54.183.26.131:8080/job/ros2_batch_ci_osx/ws/workspace/src/ros2/system_tests/test_communication/ospl-info.log

This makes it hard to debug, and it is being generated into the source space, neither of which are desirable. After doing just a little bit of research it looks like we can use `OSPL_ERRORFILE` and `OSPL_INFOFILE` to direct the location of these files. The other option is to run each executable in its own folder to prevent them from using the same file. See:

http://forums.opensplice.org/index.php?/topic/1974-log-file-management/?p=2850

The above also hints that we can build a library that gives us even more control over OpenSplice's logging and maybe the long term solution to embedding the OpenSplice logs into the ROS 2 logging system.

wjwwood, 2015-06-08:
Reading the OSPL deployment guide, it looks like `OSPL_LOGPATH` might be the right env variable now.

wjwwood, 2015-06-08:
Actually the `OSPL_ERRORFILE ` and `OSPL_INFOFILE ` env variables can be used to adjust the name, whereas `OSPL_LOGPATH` is used to be the path they are to be put into. We can also specify `<stdout>` or `<stderr>` in those env variables. We can even adjust the verbosity level with `OSPL_VERBOSITY`, though it's not clear to me what the value of that variable should be.

dirk-thomas, 2015-06-08:
Until we have more sophisticated logging options I would propose to use stdout / stderr since that output is already handled in `launch`, Jenkins as well as if invoked on the command line.

The OpenSplice package could set these environment variables in its package local setup files.

wjwwood, 2015-06-08:
I agree, I'll look at how to do that.

closed 2015-06-10 by dirk-thomas