ros2/system_tests #10: Add tests for Parameters

wjwwood, 2015-06-11:
Similar to the tests for pub/sub and services.

tfoote, 2015-06-30:
ongoing in this branch: https://github.com/ros2/system_tests/tree/test_parameters

tfoote, 2015-06-30:
Tests are running on opensplice and connext connext_dynamic is hanging.  I need to debug more to find out what's going wrong. It's only basic coverage at the moment.

gerkey, 2015-07-07:
From today's meeting: there is support in ament for: gtest, gmock, and nose.  @dirk-thomas will add a pointer to the relevant documentation.

gerkey, 2015-07-07:
Documentation is in the CMake code: https://github.com/ament/ament_cmake/blob/master/ament_cmake_gtest/cmake/ament_add_gtest.cmake

tfoote, 2015-08-22:
http://ci.ros2.org/job/ros2_batch_ci_linux/261/ pass
http://ci.ros2.org/job/ros2_batch_ci_osx/186/ pass
http://ci.ros2.org/job/ros2_batch_ci_windows/265/ (failing to find gtest. has anyone seen this error before?)

dirk-thomas, 2015-08-23:
If you look at any other Windows job you will find that GTest is always missing. Simply because GTest is not installed on the Windows slave. This will be resolved once we merge ament/ament_cmake#22

closed 2015-08-28 by tfoote