ros2/system_tests #36: timer test may be flaky

wjwwood, 2015-08-26:
Between two builds of OS X this test failed:

http://ci.ros2.org/job/ros2_batch_ci_osx/194/testReport/junit/(root)/test_time__rmw_connext_dynamic_cpp/timer_during_wait/

The previous build passed that one test.

http://ci.ros2.org/job/ros2_batch_ci_osx/193/

There were small changes which could have affected this, but I think they're unrelated since this only affects one implementation and the changes should have affected both static and dynamic connext if they were the root cause. I don't have time to dig into it right now so I created this issue.

jacquelinekay, 2015-11-04:
try to fix this week

jacquelinekay, 2015-11-05:
how ironic, I assigned this to myself but the error only seems to happen intermittently on OSX...

http://ci.ros2.org/view/nightly/job/ros2_batch_ci_osx_nightly/101/
http://ci.ros2.org/view/nightly/job/ros2_batch_ci_osx_nightly/99/

I'm now vnc'd into a Gazebo Jenkins slave because I didn't want to disrupt our Jenkins machine and Steve said it was ok...

there's also a shared_ptr_const_reference test that fails, I can look into that today.

dirk-thomas, 2015-11-05:
Windows has it too: http://ci.ros2.org/job/ros2_batch_ci_windows/565/testReport/(root)/test_time__rmw_opensplice_cpp/timer_during_wait/

closed 2015-11-06 by jacquelinekay