ros2/system_tests #37: random failure with connext

wjwwood, 2015-08-26:
Between two Linux builds which were expected to be the same this test failed:

http://ci.ros2.org/job/ros2_batch_ci_linux/274/testReport/junit/(root)/test_requester_replier_cpp__primitives__rmw_connext_dynamic_cpp__rmw_connext_cpp/test_requester_replier/

The previous build passed: http://ci.ros2.org/job/ros2_batch_ci_linux/273/

Some of the output seems to indicate a similar failure (maybe not a common cause) to this: https://community.rti.com/kb/osx510

tfoote, 2015-08-28:
This same test appears to be flaky on opensplice with a different error on osx: http://ci.ros2.org/job/ros2_batch_ci_osx/218/testReport/junit/%28root%29/test_requester_replier_cpp__primitives__rmw_opensplice_cpp/test_requester_replier/

jacquelinekay, 2015-10-15:
Duplicate of ros2/ros2#127

closed 2015-10-15 by jacquelinekay