ros2/system_tests #43: Non-primitive field types for services

ainchauspe, 2015-09-03:
Hi @dirk-thomas,

This is Anita from Erle Robotics. Recently, under the supervision of @vmayoral I've been trying to port ROS 1 packages to ROS 2. However, working with services I get errors with non-primitive field types. 
I followed these steps:

· Build the package with primitives. (In this case the package builds perfectly)

· Build the package with both primitives and non-primitives. (I get an `AttributeError: 'ServiceSpecification' object has no attribute 'base_type' ` error) 

· Again, build the package just with primitives. (I still get errors)

Here is the link of the package: https://github.com/erlerobot/ros2_spider_msgs

I'm following the Migration guide from ROS 1, am I missing something?

Thank you,

Anita

vmayoral, 2015-09-03:
Doubled checked this, it seems that services composes of non-primitive messages can't be compiled (e.g.: as soon as uncommenting https://github.com/erlerobot/ros2_spider_msgs/blob/master/srv/SrvTest.srv#L1) i get the error @ainchauspe  describes above.

Is this expected to work or nobody has looked at it yet?

dirk-thomas, 2015-09-03:
I tried to build the `spider_msgs` package and ran into a different problem. It seem that calling `rosidl_generate_interfaces` twice was the issue for me. After moving the service files into the first call of `rosidl_generate_interfaces` it worked.

It then also worked to uncomment the line you referenced.

If it still doesn't work you might want to check that you are using the latest version of all ROS 2 repos.

ainchauspe, 2015-09-04:
Thank you @dirk-thomas I made that change and now it works perfectly!

closed 2015-09-04 by ainchauspe