ros2/system_tests #50: Intermittently failing requester/replier test

jacquelinekay, 2015-10-07:
The test_requester_replier test is flaky on master.

e.g. on a recent test branch, it passed on osx but not Linux:
http://ci.ros2.org/job/ros2_batch_ci_linux/403/testReport/junit/(root)/test_requester_replier_cpp__empty__rmw_connext_dynamic_cpp/test_requester_replier/
http://ci.ros2.org/job/ros2_batch_ci_osx/300/

wjwwood, 2015-10-08:
Ok, I found the problem. I can reproduce the issue consistently with this patch:

```diff
diff --git a/test_communication/test/test_replier.cpp b/test_communication/test/test_replier.cpp
index 2ccda11..81931e3 100644
--- a/test_communication/test/test_replier.cpp
+++ b/test_communication/test/test_replier.cpp
@@ -46,6 +46,7 @@ void reply(
           std::cout << "received request #" << (index + 1) << " of " <<
             expected_services.size() << std::endl;
           *response = *expected_service.second;
+          rclcpp::sleep_for(100_ms);
           known_request = true;
           if (received_messages[index]) {
             fprintf(stderr, "received the same request multiple times\n");

```

Basically what's happening is that the requester sends a request and then waits up to 100ms for the response. If it takes longer than 100ms for the response it discards the future and tries to send the request again. In the mean time the replier has received both requests and sent a reply for the first but not in time for the requester to get the reply. So as far as the replier is concerned the requester should have gotten the reply and moved on to the next request, but instead it sends the first request a second time and the replier sees this as an error which is why you get the `received the same request multiple times`.

By introducing the sleep inside the replier, I forced the same condition to always be true. I'm still looking at how to fix this in a reliable way. I think it might be appropriate for the replier to just ignore duplicate requests from the requester.

dirk-thomas, 2015-10-08:
I think the tests for these duplicates for an important reason: if a client "drops" the future of a previous request I don't think he should be notified later. If the response still arrives I think discarding it instead of dispatching it is a more reasonable behavior.

wjwwood, 2015-10-08:
@dirk-thomas I don't understand what you're saying.

> if a client "drops" the future of a previous request I don't think he should be notified later.

If the client discards a future for a given request, they are not notified, since they never check the future.

I don't follow what you're arguing for nor what we should do instead.

dirk-thomas, 2015-10-08:
Nvm. I thought the check for duplicates was on the client side.

wjwwood, 2015-10-08:
Yeah, this is checking on the server side.

PR #51 (2015-10-09): disregard duplicate requests
fixes #50
commit ed93a9f6 (2015-10-09): make duplicate requests just a warning not a failure

fixes #50
closed 2015-10-09 by wjwwood