ros2/system_tests #93: Services test is flaky

gerkey, 2015-12-17:
Starting from a failure in a nightly job (http://ci.ros2.org/view/nightly/job/nightly_osx/147/consoleFull#console-section-170 - search for `test_services_cpp__rmw_opensplice_cpp ..........................***Timeout`), I can reproduce an apparent deadlock in the services test like so:

~~~
while true; do nosetests3 -s ./build/test_rclcpp/test_services_cpp__rmw_opensplice_cpp.py; done
~~~

Eventually, it'll stop. Then if I attach gdb to each process, the server has this backtrace:

~~~
#0  0x0000000000416127 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7ffeb6060268, 
    __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:545
#1  0x0000000000415fc6 in std::__shared_ptr<rclcpp::node::Node, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7ffeb6060260, 
    __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:781
#2  0x0000000000415fe0 in std::shared_ptr<rclcpp::node::Node>::~shared_ptr (this=0x7ffeb6060260, __in_chrg=<optimized out>)
    at /usr/include/c++/4.8/bits/shared_ptr.h:93
#3  0x00007ff01409578a in rclcpp::executor::Executor::get_next_timer(std::shared_ptr<rclcpp::executor::AnyExecutable>) ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#4  0x00007ff014095c25 in rclcpp::executor::Executor::get_next_ready_executable() ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#5  0x00007ff014095eda in rclcpp::executor::Executor::get_next_executable(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
    () from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#6  0x00007ff01409dd5b in rclcpp::executors::single_threaded_executor::SingleThreadedExecutor::spin() ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#7  0x00007ff01409928c in rclcpp::spin(std::shared_ptr<rclcpp::node::Node>) ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#8  0x0000000000415d8d in main (argc=1, argv=0x7ffeb60605e8)
    at /home/gerkey/ros2_ws/src/ros2/system_tests/test_rclcpp/test/test_services_server.cpp:50
~~~

Not surprising; it's waiting in `spin()`.

The client's backtrace is:

~~~
#0  0x0000000000489fce in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::~_Sp_counted_base (this=0xff0d00, 
    __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:114
#1  0x000000000049311f in std::_Sp_counted_ptr<test_rclcpp::srv::AddTwoInts_Response_<std::allocator<void> >*, (__gnu_cxx::_Lock_policy)2>::~_Sp_counted_ptr (this=0xff0d00, __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:281
#2  0x000000000049314e in std::_Sp_counted_ptr<test_rclcpp::srv::AddTwoInts_Response_<std::allocator<void> >*, (__gnu_cxx::_Lock_policy)2>::~_Sp_counted_ptr (this=0xff0d00, __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:281
#3  0x00000000004931ed in std::_Sp_counted_ptr<test_rclcpp::srv::AddTwoInts_Response_<std::allocator<void> >*, (__gnu_cxx::_Lock_policy)2>::_M_destroy (this=0xff0d00) at /usr/include/c++/4.8/bits/shared_ptr_base.h:294
#4  0x0000000000487520 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xff0d00)
    at /usr/include/c++/4.8/bits/shared_ptr_base.h:161
#5  0x00000000004860d3 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7ffd77070218, 
    __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:546
#6  0x00007f65a1fc0780 in std::__shared_ptr<void, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#7  0x00007f65a1fc079a in std::shared_ptr<void>::~shared_ptr() ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#8  0x00007f65a1fc28f6 in rclcpp::executor::Executor::execute_client(std::shared_ptr<rclcpp::client::ClientBase>) ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#9  0x00007f65a1fc21c4 in rclcpp::executor::Executor::execute_any_executable(std::shared_ptr<rclcpp::executor::AnyExecutable>) ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#10 0x00007f65a1fc1d52 in rclcpp::executor::Executor::spin_once(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp__rmw_opensplice_cpp.so
#11 0x0000000000489ee0 in rclcpp::executor::Executor::spin_until_future_complete<std::pair<std::shared_ptr<test_rclcpp::srv::AddTwoInts_Request_<std::allocator<void> > >, std::shared_ptr<test_rclcpp::srv::AddTwoInts_Response_<std::allocator<void> > > >, std::ratio<1l, 1000l> > (this=0x7ffd77070400, future=..., timeout=...) at /home/gerkey/ros2_ws/install/include/rclcpp/executor.hpp:167
#12 0x0000000000488d30 in rclcpp::executors::spin_node_until_future_complete<std::pair<std::shared_ptr<test_rclcpp::srv::AddTwoInts_Request_<std::allocator<void> > >, std::shared_ptr<test_rclcpp::srv::AddTwoInts_Response_<std::allocator<void> > > >, std::ratio<1l, 1000l> > (executor=..., node_ptr=..., future=..., timeout=...) at /home/gerkey/ros2_ws/install/include/rclcpp/executors.hpp:67
#13 0x0000000000487440 in rclcpp::spin_until_future_complete<std::pair<std::shared_ptr<test_rclcpp::srv::AddTwoInts_Request_<std::allocator<void> > >, std::shared_ptr<test_rclcpp::srv::AddTwoInts_Response_<std::allocator<void> > > >, std::ratio<1l, 1000l> > (
    node_ptr=..., future=..., timeout=...) at /home/gerkey/ros2_ws/install/include/rclcpp/executors.hpp:81
#14 0x00000000004841d1 in test_services_client_test_return_request_Test::TestBody (this=0xf88e40)
    at /home/gerkey/ros2_ws/src/ros2/system_tests/test_rclcpp/test/test_services_client.cpp:76
#15 0x00000000004aeddc in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0xf88e40, 
    method=&virtual testing::Test::TestBody(), location=0x4bb49b "the test body") at /usr/src/gtest/src/gtest.cc:2090
#16 0x00000000004aa3cc in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0xf88e40, 
    method=&virtual testing::Test::TestBody(), location=0x4bb49b "the test body") at /usr/src/gtest/src/gtest.cc:2126
#17 0x0000000000497e83 in testing::Test::Run (this=0xf88e40) at /usr/src/gtest/src/gtest.cc:2162
#18 0x0000000000498588 in testing::TestInfo::Run (this=0xf88830) at /usr/src/gtest/src/gtest.cc:2338
#19 0x0000000000498ae4 in testing::TestCase::Run (this=0xf885d0) at /usr/src/gtest/src/gtest.cc:2445
#20 0x000000000049d58a in testing::internal::UnitTestImpl::RunAllTests (this=0xf882d0) at /usr/src/gtest/src/gtest.cc:4243
#21 0x00000000004afd1d in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (
    object=0xf882d0, 
    method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x49d31c <testing::internal::UnitTestImpl::RunAllTests()>, location=0x4bbfd8 "auxiliary test code (environments or event listeners)") at /usr/src/gtest/src/gtest.cc:2090
#22 0x00000000004ab2d6 in testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (
    object=0xf882d0, 
    method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x49d31c <testing::internal::UnitTestImpl::RunAllTests()>, location=0x4bbfd8 "auxiliary test code (environments or event listeners)") at /usr/src/gtest/src/gtest.cc:2126
#23 0x000000000049c4f5 in testing::UnitTest::Run (this=0x6dfc00 <testing::UnitTest::GetInstance()::instance>)
    at /usr/src/gtest/src/gtest.cc:3880
#24 0x00000000004842f4 in main (argc=2, argv=0x7ffd77070938)
    at /home/gerkey/ros2_ws/src/ros2/system_tests/test_rclcpp/test/test_services_client.cpp:83
~~~

It's waiting in `spin_until_future_complete()` for the `test_return_request` test (https://github.com/ros2/system_tests/blob/master/test_rclcpp/test/test_services_client.cpp#L70 - line numbers are different in my backtrace because I'm adding print statements).

From added print statements, I can see that the server gets the callback `handle_add_two_ints_noreqid()` invoked once with args `(1,2)` and gets the callback `handle_add_two_ints_reqid()` invoked twice, both times with args `(4,5)`. So all three service requests are getting to the server. But the third response doesn't make its way back to the client.

gerkey, 2015-12-17:
@esteve, any thoughts on this problem?

esteve, 2015-12-17:
@jacquelinekay suggested that it might be because the service gets killed too early, so it won't send the response back to the client in time. Looking into it.

gerkey, 2015-12-17:
What do you mean by the service getting killed? In this case, both processes (server and client) are still running when the system is deadlocked, and the server appears to have happily serviced all the requests and then gone back to spinning.

esteve, 2015-12-17:
@gerkey sorry, I thought that was different issue.

jacquelinekay, 2015-12-18:
Potential fix in #95. I was getting deadlock after ~20 iterations on my desktop and I don't see it anymore.

PR #95 (2015-12-18): Try to fix flaky services test by partitioning topic names
Fixes #93

I think there was a problem with crosstalk in DDS entities that weren't finished tearing down. I've separated all nodes and service names for the different cases.

The test is turning over reliably on my machine (has been for over 20 minutes). I'll start a repeat-until-fail job with this test on OSX, since that's the platform where it failed in the nightly job.
closed 2015-12-18 by jacquelinekay