ros2/system_tests #134: Test fails with extra request

gerkey, 2016-06-14:
Example:

http://ci.ros2.org/view/nightly/job/nightly_linux_debug/105/testReport/(root)/test_requester_replier_cpp__Primitives__rmw_opensplice_cpp_Debug/test_requester_replier/

I can reproduce this problem locally. `test_replier` receives the 2 expected requests, which looks like this:
```
$5 = (test_communication::srv::Primitives_Request_<std::allocator<void> > &) @0x6949d0: {bool_value = false, byte_value = 0 '\000', 
  char_value = 0 '\000', float32_value = 0, float64_value = 0, int8_value = 0 '\000', uint8_value = 0 '\000', int16_value = 0, 
  uint16_value = 0, int32_value = 0, uint32_value = 0, int64_value = 0, uint64_value = 0, string_value = "request"}
```
```
$6 = (test_communication::srv::Primitives_Request_<std::allocator<void> > &) @0x6949d0: {bool_value = false, byte_value = 0 '\000', 
  char_value = 0 '\000', float32_value = 0, float64_value = 0, int8_value = 0 '\000', uint8_value = 0 '\000', int16_value = 0, 
  uint16_value = 0, int32_value = 0, uint32_value = 0, int64_value = 0, uint64_value = 0, string_value = "request"}
```
and then receives a third, unexpected one, which looks bogus:
```
$8 = (test_communication::srv::Primitives_Request_<std::allocator<void> > &) @0x68eb70: {bool_value = false, byte_value = 214 '\326', 
  char_value = -1 '\377', float32_value = 4.59163468e-41, float64_value = 6.953355807310109e-310, int8_value = 112 'p', 
  uint8_value = 214 '\326', int16_value = -1, uint16_value = 32767, int32_value = -10608, uint32_value = 32767, 
  int64_value = 140737488344768, uint64_value = 6969568, string_value = ""}
```

wjwwood, 2016-06-14:
I think I fixed this in my not yet merged pr: https://github.com/ros2/rmw_opensplice/commit/1a73e14fe868ef8a845b53cfa5c88a7d32fa4cfb

gerkey, 2016-06-14:
Glad I asked! I'll test with that change.

gerkey, 2016-06-14:
Problem fixed with ros2/rmw_opensplice@1a73e14 applied locally. Thanks for the tip. I'll leave this issue open until your PR gets merged, in case somebody else comes across the problem.

dhood, 2016-09-21:
http://ci.ros2.org/job/ci_linux/1736/testReport/ confirms that this was fixed by https://github.com/ros2/rmw_opensplice/pull/143

closed 2016-09-21 by dhood