ros2/system_tests #143: remove disabled tests when wait_for_service support is added to fastrtps

wjwwood, 2016-06-23:
See: https://github.com/ros2/ros2/issues/215#issuecomment-226589984

Links:
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-d9905b3caf2478827ea45e9627bca66cR19
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-d9905b3caf2478827ea45e9627bca66cR24
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-d9905b3caf2478827ea45e9627bca66cR42
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-7a4fbef276d5cd364fc8902acf056c6cR17
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-7a4fbef276d5cd364fc8902acf056c6cR40
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-31e1e68750ca0dfea0e0bb6371955114R16
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-31e1e68750ca0dfea0e0bb6371955114R33
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-2405ee84fb93bcb417d75a061766aadbR20
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-2405ee84fb93bcb417d75a061766aadbR208
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-af0063cf5fa30069b9fb91b23db82ea2R16
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-af0063cf5fa30069b9fb91b23db82ea2R49
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-af0063cf5fa30069b9fb91b23db82ea2R115
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-b3fa04db905d6df0ea534488541c8173R17
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-b3fa04db905d6df0ea534488541c8173R197
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-b3fa04db905d6df0ea534488541c8173R226
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-a9e2ae6b2150d00377652d9d92752f04R16
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-a9e2ae6b2150d00377652d9d92752f04R41
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-a9e2ae6b2150d00377652d9d92752f04R65
- https://github.com/ros2/system_tests/commit/a66ae7198ab00435dcbe11a9e6a991eaed4b94f2#diff-a9e2ae6b2150d00377652d9d92752f04R90

wjwwood, 2016-06-23:
Tracked: eProsima/ROS-RMW-Fast-RTPS-cpp#14

mikaelarguedas, 2017-08-05:
closing this given that it seems the have been adressed in https://github.com/ros2/system_tests/pull/164

closed 2017-08-05 by mikaelarguedas