ros2/system_tests #153: two clients with the same settings behave differently for fastrtps

dhood, 2016-07-07:
In https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/issues/48 (`test_client_scope_cpp` failures), the first client received the response but the second client did not. It can be fixed by making a change to the QoS reliability settings of the second client.

The second client in that demo should have the same behaviour as the first. I have made a regression test that currently fails for fastrtps: https://github.com/ros2/system_tests/commit/0b024f50539643f9840303c0a0dc70afe6b26eb2

it can be fixed by setting
```
rmw_qos_profile.reliability = RMW_QOS_POLICY_RELIABILITY_SYSTEM_DEFAULT;
```
for the second client.

should I open a PR even though the test is failing? or is this ticket sufficient?

wjwwood, 2016-07-08:
If you've committed the test to a branch, then you might as well open a pull request so it's more visible. Otherwise, it's not clear why the branch is there and if it can be pruned or not in the future.

dhood, 2016-07-09:
thanks, I've done that now

dhood, 2016-09-21:
closed with https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/issues/48#issuecomment-236144061
regression test added in https://github.com/ros2/system_tests/issues/156

closed 2016-09-21 by dhood