ros2/system_tests #154: possible test-result masking

gerkey, 2016-07-08:
While working on ros2/rclcpp#236, I ran this build: http://ci.ros2.org/job/ci_osx/1232/testReport/, which shows some fastrtps test failures. Looking at the console log (http://ci.ros2.org/job/ci_osx/1232/consoleFull#console-section-200), among much other output, I noticed this section:
```
23:52:30 test 13
23:52:30       Start 13: test_parameter_server_cpp__rmw_connext_cpp
23:52:30 
23:52:30 13: Test command: /Users/osrf/jenkins/workspace/ci_osx/venv/bin/python3 "-u" "/Users/osrf/jenkins/workspace/ci_osx/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/Users/osrf/jenkins/workspace/ci_osx/ws/build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_connext_cpp.xunit.xml" "--output-file" "/Users/osrf/jenkins/workspace/ci_osx/ws/build/test_rclcpp/ament_cmake_nose/test_parameter_server_cpp__rmw_connext_cpp.txt" "--env" "RCL_ASSERT_RMW_ID_MATCHES=rmw_connext_cpp" "--command" "/Users/osrf/jenkins/workspace/ci_osx/venv/bin/python3" "/Users/osrf/jenkins/workspace/ci_osx/venv/bin/nosetests" "/Users/osrf/jenkins/workspace/ci_osx/ws/build/test_rclcpp/test_parameter_server_cpp__rmw_connext_cpp_.py" "--with-xunit" "--xunit-file=/Users/osrf/jenkins/workspace/ci_osx/ws/build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_connext_cpp.xunit.xml" "--xunit-testsuite-name=test_rclcpp.nosetests"
23:52:30 13: Test timeout computed to be: 15
23:52:30 13: -- run_test.py: extra environment variables:
23:52:30 13:  - RCL_ASSERT_RMW_ID_MATCHES=rmw_connext_cpp
23:52:30 13: -- run_test.py: invoking following command in '/Users/osrf/jenkins/workspace/ci_osx/ws/src/ros2/system_tests/test_rclcpp':
23:52:30 13:  - /Users/osrf/jenkins/workspace/ci_osx/venv/bin/python3 /Users/osrf/jenkins/workspace/ci_osx/venv/bin/nosetests /Users/osrf/jenkins/workspace/ci_osx/ws/build/test_rclcpp/test_parameter_server_cpp__rmw_connext_cpp_.py --with-xunit --xunit-file=/Users/osrf/jenkins/workspace/ci_osx/ws/build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_connext_cpp.xunit.xml --xunit-testsuite-name=test_rclcpp.nosetests
23:52:45 13/35 Test #13: test_parameter_server_cpp__rmw_connext_cpp .................***Timeout  15.09 sec
```

That sure looks to me like the connext variant of the parameter_server test failed. But there's no report of any connext failures in Jenkins. I wonder whether the fastrtps and connext test variants are masking each other in the report in one direction or the other. In this case, the fastrtps variant of the parameter_server test also timed out and it was reported as a failure.

dhood, 2016-07-08:
I think the test was then re-run and passed (go to the last occurrence of `test_parameter_server_cpp__rmw_connext_cpp` in the log)

gerkey, 2016-07-08:
Ah, brilliant! I was expecting the re-run to happen immediately if it was enabled, and didn't look far enough ahead to find the `23:54:29 +++ Testing 'test_rclcpp' again (retry #1 of 5)` output. Thanks.

closed 2016-07-08 by gerkey