ros2/system_tests #191: Move messages and fixtures from test_communication into another repo

gerkey, 2017-03-10:
ros2/system_tests#190 arranges for installation of messages and helpful Python code that's now being used elsewhere (ros2/cli_tools#5). But it would be better to move that stuff into a different repo (https://github.com/ros2/system_tests/pull/190#issuecomment-285801282).

mikaelarguedas, 2017-04-05:
see https://github.com/ros2/rcl_interfaces/pull/16#issue-219022063 for what remains to be done

mikaelarguedas, 2017-04-26:
@gerkey is this still on your todo list ?

gerkey, 2017-04-26:
It's certainly not near the top of my stack. If it becomes urgent, I'm happy to work on it.

mikaelarguedas, 2017-04-26:
Should we unassign this until someone that has time picks it up ?

dirk-thomas, 2017-04-26:
We have already copied the messages / fixtures to a second location. Therefore in my opinion we should follow up on also using them from here and not build up more unnecessary technical debt.

mikaelarguedas, 2017-05-01:
> Therefore in my opinion we should follow up on also using them from here

:+1: this will allow to reduce the amount of code to maintain and build time to not have to build messages for every package needing to test anything message related, so I also think that we should follow-up on this sooner than later.

dirk-thomas, 2017-05-16:
@gerkey We assigned this ticket two weeks ago for the sprint. Is there any update on this?

mikaelarguedas, 2017-09-01:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2017-09-01:
Ok I finally found some time to move this forward.
What this set of PR does:
- move interfaces and corresponding C++/Python fixtures into a new package called test_msgs
- update packages depending on test_communication to use this package
- test_communication is now a real testing package that doesn't need to install anything (or to be released :wink: @nuclearsandwich )

What is not in this set of PRs:
- messages that don't have fixtures are still generated by this package
- no fancy cmake tricks to get the list of messages that have fixtures because they all have some so I'm just relying on the ament_index
- no refactor of any test whatsoever
- didn't modify other packages to use these messages rather than generating their own messages for testing (thinking rclcpp, test_rclcpp)

closed 2017-09-08 by mikaelarguedas