ros2/system_tests #195: how to run tests

k-okada, 2017-03-19:
Sorry for basic question, how to run tests?
I'm using beta1 release and ` ament test ~/ros2_ws/src/ros2/system_tests/test_communication/` outputs many failures.

```

38: [test_subscriber] exception in subscriber:
38: [test_subscriber] Traceback (most recent call last):
38: [test_subscriber]   File "/home/user/ros2_ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 88, in <module>
38: [test_subscriber]     listener(message_name=args.message_name)
38: [test_subscriber]   File "/home/user/ros2_ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 69, in listener
38: [test_subscriber]     qos_profile_default)
38: [test_subscriber]   File "/home/user/ros2_ws/install/lib/python3.5/site-packages/rclpy/node.py", line 62, in create_subscription
38: [test_subscriber]     raise NoTypeSupportImportedException
38: [test_subscriber] rclpy.exceptions.NoTypeSupportImportedException: no type_support imported
38: (test_subscriber) rc 1
38: () tear down
38: (test_publisher) signal SIGINT
38: [test_publisher] signal_handler(2)
38: [test_publisher] published for 0.156372 seconds
38: (test_publisher) rc 0

The following tests FAILED:
         13 - test_publisher_subscriber__BoundedArrayNested__rclpy__rmw_fastrtps_cpp (Failed)
         14 - test_publisher_subscriber__BoundedArrayPrimitives__rclpy__rmw_fastrtps_cpp (Failed)
         15 - test_publisher_subscriber__Builtins__rclpy__rmw_fastrtps_cpp (Failed)
         16 - test_publisher_subscriber__DynamicArrayNested__rclpy__rmw_fastrtps_cpp (Failed)
         17 - test_publisher_subscriber__DynamicArrayPrimitives__rclpy__rmw_fastrtps_cpp (Failed)
         18 - test_publisher_subscriber__Empty__rclpy__rmw_fastrtps_cpp (Failed)
         19 - test_publisher_subscriber__Nested__rclpy__rmw_fastrtps_cpp (Failed)
         20 - test_publisher_subscriber__Primitives__rclpy__rmw_fastrtps_cpp (Failed)
         21 - test_publisher_subscriber__StaticArrayNested__rclpy__rmw_fastrtps_cpp (Failed)
         22 - test_publisher_subscriber__StaticArrayPrimitives__rclpy__rmw_fastrtps_cpp (Failed)
         23 - test_requester_replier__Empty__rclpy__rmw_fastrtps_cpp (Failed)
         24 - test_requester_replier__Primitives__rclpy__rmw_fastrtps_cpp (Failed)
         25 - test_publisher_subscriber__BoundedArrayNested__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         26 - test_publisher_subscriber__BoundedArrayPrimitives__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         27 - test_publisher_subscriber__Builtins__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         28 - test_publisher_subscriber__DynamicArrayNested__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         29 - test_publisher_subscriber__DynamicArrayPrimitives__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         30 - test_publisher_subscriber__Empty__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         31 - test_publisher_subscriber__Nested__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         32 - test_publisher_subscriber__Primitives__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         33 - test_publisher_subscriber__StaticArrayNested__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         34 - test_publisher_subscriber__StaticArrayPrimitives__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         35 - test_requester_replier__Empty__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         36 - test_requester_replier__Primitives__rclpy__rclcpp__rmw_fastrtps_cpp (Failed)
         37 - test_publisher_subscriber__BoundedArrayNested__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         38 - test_publisher_subscriber__BoundedArrayPrimitives__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         39 - test_publisher_subscriber__Builtins__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         40 - test_publisher_subscriber__DynamicArrayNested__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         41 - test_publisher_subscriber__DynamicArrayPrimitives__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         42 - test_publisher_subscriber__Empty__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         43 - test_publisher_subscriber__Nested__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         44 - test_publisher_subscriber__Primitives__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         45 - test_publisher_subscriber__StaticArrayNested__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         46 - test_publisher_subscriber__StaticArrayPrimitives__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         47 - test_requester_replier__Empty__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
         48 - test_requester_replier__Primitives__rclcpp__rclpy__rmw_fastrtps_cpp (Failed)
```

mikaelarguedas, 2017-03-20:
this is the right way to run the tests (alternatively you can pass `--only PACKAGE_NAME` rather than the full path of your package).

Looking at the failure, it looks like only the tests involving python are failing. The error message seems to indicate that `rclpy` couldn't find fastrtps when creating subscribers/publishers.
are you able to run `talker_py` and `listener_py` after sourcing this workspace?

k-okada, 2017-03-21:
Yes, python example working ok, full logs are here, I have to change Fast-RTPS for https://github.com/eProsima/Fast-RTPS/issues/77
```
Script started on Tue 21 Mar 2017 05:09:09 AM UTC
user@ubuntu-16:~/ros2_ws/src$ wstool info
workspace: /home/user/ros2_ws/src

 Localname               S SCM Version (Spec)                                         UID  (Spec)                 URI  (Spec) [http(s)://...]
 ---------               - --- --------------                                         -----------                 ---------------------------
 vendor/console_bridge     git <detached>  (148df7e841b91da488ec3bb4abd295a2bccdb728) 148df7e841b9                github.com/ros/console_bridge.git
 ros2/tlsf                 git <detached>  (release-beta1)                            69af8ace93b0                github.com/ros2/tlsf.git
 ros2/system_tests       V git add_mutibyte_tests  (release-beta1)                    e2c2a2f81d40 (c0b9c39f9d72) github.com/ros2/system_tests.git
 ros2/rosidl_typesupport   git <detached>  (release-beta1)                            b0e330cc592f                github.com/ros2/rosidl_typesupport.git
 ros2/rosidl_dds           git <detached>  (release-beta1)                            bfd65a903487                github.com/ros2/rosidl_dds.git
 ros2/rosidl             V git add_multibyte  (release-beta1)                         d075437bac0d (a3074a616e43) github.com/ros2/rosidl.git
 ros2/ros1_bridge          git <detached>  (release-beta1)                            5de8656614ab                github.com/ros2/ros1_bridge.git
 ros2/rmw_opensplice       git <detached>  (release-beta1)                            853aba13aee4                github.com/ros2/rmw_opensplice.git
 ros2/rmw_implementation   git <detached>  (release-beta1)                            4e9f8a3d8430                github.com/ros2/rmw_implementation.git
 ros2/rmw_fastrtps         git <detached>  (release-beta1)                            f9fc9a1b9874                github.com/ros2/rmw_fastrtps.git
 ros2/rmw_connext          git <detached>  (release-beta1)                            5449e8d1484b                github.com/ros2/rmw_connext.git
 ros2/rmw                  git <detached>  (release-beta1)                            18b08d89a118                github.com/ros2/rmw.git
 ros2/realtime_support     git <detached>  (release-beta1)                            8e2b486c2b99                github.com/ros2/realtime_support.git
 ros2/rclpy                git <detached>  (release-beta1)                            5209e5e95add                github.com/ros2/rclpy.git
 ros2/rclcpp               git <detached>  (release-beta1)                            d00a441195e8                github.com/ros2/rclcpp.git
 ros2/rcl_interfaces       git <detached>  (release-beta1)                            1e6202ab6863                github.com/ros2/rcl_interfaces.git
 ros2/rcl                  git <detached>  (release-beta1)                            638f976783ca                github.com/ros2/rcl.git
 ros/poco_vendor           git <detached>  (release-beta1)                            00646ef80ea2                github.com/ros2/poco_vendor.git
 ros2/launch               git <detached>  (release-beta1)                            21315578f2f0                github.com/ros2/launch.git
 ros2/examples             git <detached>  (release-beta1)                            b57f9591d964                github.com/ros2/examples.git
 ros2/demos                git <detached>  (release-beta1)                            02178ac928f8                github.com/ros2/demos.git
 ros2/common_interfaces    git <detached>  (release-beta1)                            4872c6e0c479                github.com/ros2/common_interfaces.git
 ros/geometry2             git <detached>  (3a820ee7238e406282c901991186adc6558cdb5e) 3a820ee7238e                github.com/ros/geometry2.git
 ros/class_loader          git <detached>  (a7e9af992f9fb3ba6b859203a322153ee82f4de5) a7e9af992f9f                github.com/ros/class_loader.git
 eProsima/Fast-RTPS      M git <detached>  (0a141ba330c65b39985c1ee3823edeb0e38b25ca) 0a141ba330c6                github.com/eProsima/Fast-RTPS.git
 eProsima/Fast-CDR         git <detached>  (8653f20bd3028621cc6b2f06fb6e0477d5f120fc) 8653f20bd302                github.com/eProsima/Fast-CDR.git
 ament/uncrustify          git <detached>  (release-beta1)                            0cb6cf0f5c59                github.com/ament/uncrustify.git
 ament/osrf_pycommon       git <detached>  (8aaf12c5538ddc1aa5a6eba912c518e4e68f717f) 8aaf12c5538d                github.com/osrf/osrf_pycommon.git
 ament/gtest_vendor        git <detached>  (release-beta1)                            20fe2efdbdcb                github.com/ament/gtest_vendor.git
 ament/gmock_vendor        git <detached>  (release-beta1)                            c0a41c28ac4f                github.com/ament/gmock_vendor.git
 ament/ament_tools         git <detached>  (release-beta1)                            5811a5002233                github.com/ament/ament_tools.git
 ament/ament_package       git <detached>  (release-beta1)                            778906c805fd                github.com/ament/ament_package.git
 ament/ament_lint          git <detached>  (release-beta1)                            8f5a186631a0                github.com/ament/ament_lint.git
[... cut ...]
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.5/subprocess.py", line 559, in call
    return p.wait(timeout=timeout)
  File "/usr/lib/python3.5/subprocess.py", line 1658, in wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.5/subprocess.py", line 1608, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt
Makefile:93: recipe for target 'test' failed
make: *** [test] Interrupt
user@ubuntu-16:~/ros2_ws/src$ exit
exit

Script done on Tue 21 Mar 2017 05:09:52 AM UTC
```

dirk-thomas, 2017-07-10:
@k-okada Sorry for the late reply.

Instead of passing the path to the package to test to `ament test` you should call `ament test --only <pkgname>`. The first approach will only consider the single package and it is your responsibility to ensure that all dependencies are available and the environment is setup correctly. In the second approach the tool will make sure that all dependencies within this workspace are automatically sourced before running the tests. In both cases you need to make sure that the package in question as well as all dependencies have been built before.

Please comment with more information if the ticket is still relevant or close it if it has been resolved in the meantime.

k-okada, 2017-07-11:
The title of this issue is not precise, I want to ask "I dit follow the instruction of  beta1 release and try to run ament test, but it fails, Is it expected behavior ? or did I misunderstand something? for example the way to run ament test?"
And it turns out I did correct way to run test, "ament test " was ok. The problem was there is something wrong if we want run RTPS on droplet virtual machine, https://github.com/eProsima/Fast-RTPS/issues/77

closed 2017-07-11 by k-okada