ros2/system_tests #212: New failing test: test_client_scope_cpp__rmw_connext_cpp

clalancette, 2017-06-27:
Starting with build: http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/696/ , the test known as test_client_scope_cpp__rmw_connext_cpp started failing.  If you search for 'test_client_scope_cpp' in the console output, you'll see that this test started timing out.

mikaelarguedas, 2017-06-27:
known to be flaky

dhood, 2018-03-01:
closing as duplicate of https://github.com/ros2/build_cop/issues/41

dhood, 2018-03-01:
actually I think I might have been incorrect that it was tracked by the other ticket, it seems that https://github.com/ros2/build_cop/issues/41 only refers to fastrtps. services are known to take up to 20s on connext, so maybe increasing the timeout is enough for that test (it starts two clients). I'll check if that makes a difference.

closed 2018-03-09 by dhood