ros2/system_tests #238: [test_security] test access control policies

mikaelarguedas, 2017-10-13:
This is tracking the parts of `test_security` not implemented in #230 
#230 tests authentication and encryption in various scenarios but doesnt test access control.

Tests for access control should:
- have a yaml file (same format as [the sros2 example](https://github.com/ros2/sros2/blob/b4325d83ef5b9d57b025da00333b1f6179175675/examples/sample_policy.yaml)) defining all the access control policies used by the tests
- use the `ros2 security` tool to generate keys, certificates, governance and permission files for these tests
- test successful communication between nodes respecting the predefined access control policies
- test access denial for nodes violating the permissions defined in the yaml policies files

ruffsl, 2018-02-22:
@mikaelarguedas , I was planning on implementing something like this soon. Perhaps we could check the correctness of the implementation by exhaustively testing the access control system given a global policy definition.

Simply, we could manifest a policy definition into what the transport expects, synthesize a ROS graph representation (perhaps adversely every node tries to connect with all other nodes via any subsystem), then attempt to realize the graph representation using the transport. By logging or auditing the allowed and denied events, we should assert that the two sets of allowed and denied connections are equivalent to those that could be generated by evaluating the graph representation against the policy definition statically.

In detail, perhaps the *graph representation* could be formalized by constructing a bipartite graph between subjects (ros nodes, as namespaced) and objects (ros subsitems, like topics/services/etc..), and so emulating the topology of an arbitrary ROS2 graph. We could then generate/load *graph representations* that may or may not be completely comply semantically (i.e. erlevent of the transport) with the global *policy definition*. We then feed the edges from a *graph representation* through the *policy definition* to generate a synthetic labeled set of allowed and denied connections.

Next we could generate transport specific artifacts that manifest the global *policy definition*, relative for each subject, then autidit the outcome of deploying the *graph representation* by using the transport of choice. This could be done all at once or more resably via pairwise enumeration. Then we simply compare the allow/deny sets from the synthetic and transport test cases to ensure they are equivalent. From there we could easily classify any false positives or negatives in the transport enforcement of the policy.

As for any soundness or completeness, I'm not sure we'd ever be able to guarantee either if the policy permission are formulation of complex regular expressions rather than exact strings. But for exact string expressions, finite sized graphs, and a representation that provides global test coverage for a given policy (and all permissions entailed), this might provided a reasonable exhaustive benchmark.

ruffsl, 2018-02-22:
With regards to realizing a graph representation into real ROS2 nodes, which test example should I look at for programmatically testing and creating nodes and subsystems?