ros2/system_tests #242: Security "not connecting" tests are flaky

dhood, 2017-10-16:
http://ci.ros2.org/job/nightly_osx_repeated/863

```
projectroot.test_secure_publisher_subscriber__DynamicArrayNested__rmw_connext_cpp__secure_not_connecting_0
projectroot.test_secure_publisher_subscriber__DynamicArrayNested__rmw_connext_cpp__secure_not_connecting_1
projectroot.test_secure_publisher_subscriber__Empty__rmw_connext_cpp__secure_not_connecting_1
projectroot.test_secure_publisher_subscriber__DynamicArrayNested__rmw_fastrtps_cpp__secure_not_connecting_1
```

They are receiving connections when they're not supposed to. We know that some nodes can take a while to shutdown properly so perhaps it's cross-talk from re-running the tests multiple times. I'll add a namespace to the nodes and see if it fixes it.

mikaelarguedas, 2017-10-16:
I'm afraid adding namespaces is not an easy fix in this case because while the "not connecting" ones won't connect, we would need to generate different security files for all the ones that "should connect" as their current permissions have the partition "rt_".

Edit: Actually looking at the current permission files they all have wildcards on partitions so namespaces should be ok for now :+1:  let's try it

dhood, 2017-10-17:
closed by https://github.com/ros2/system_tests/pull/243

closed 2017-10-17 by dhood