ros2/system_tests #290: [test_communication] memory leak issues in test_messages_c.cpp

yechun1, 2018-08-15:
1.  A non-heap pointer is placed on the free list, likely causing a crash later.
In verify_message<test_msgs__msg__StaticArrayPrimitives>(T1 &, unsigned long): Free of an address-of expression, which can never be heap allocated ([CWE-590](http://cwe.mitre.org/data/definitions/590.html))
  
   * Input expected_msg is struct:

> ros2/system_tests/test_communication/test/test_messages_c.cpp:496
> 495  test_msgs__msg__StaticArrayPrimitives expected_msg;
> 496: get_message(&expected_msg, msg_num); 

> install/test_msgs/include/test_msgs/msg/static_array_primitives__struct.h:21
> 21 typedef struct test_msgs__msg__StaticArrayPrimitives
   * trace functions the msg be freed

> build/test_msgs/rosidl_generator_c/test_msgs/msg/static_array_primitives__functions.c:
> 86 test_msgs__msg__StaticArrayPrimitives__destroy(test_msgs__msg__StaticArrayPrimitives * msg)
> 87 {
> 88  if (msg) {
> 89    test_msgs__msg__StaticArrayPrimitives__fini(msg);
> 90  }
> 91  free(msg);
> 92 }

2. Incorrect values may be read from freed memory, causing incorrect computations.
In get_message<test_msgs__msg__StaticArrayPrimitives>(T1 *, unsigned long): Reads target of a freed pointer ([CWE-416](http://cwe.mitre.org/data/definitions/416.html))
> ros2/system_tests/test_communication/test/test_messages_c.cpp:447
> 443 void get_message(test_msgs__msg__StaticArrayPrimitives * msg, size_t msg_num)
> 444 {
> // may freed in test_msgs__msg__StaticArrayPrimitives__destroy when init
> 445  test_msgs__msg__StaticArrayPrimitives__init(msg) 
> 446  if (msg_num == 0) 
> 447    msg->bool_values[0] = false;

sloretz, 2018-08-16:
Thanks for the issue @yechun1. Mind providing steps to reproduce the back trace?

Looking at the second bullet point I'm not where is the stack pointer being free'd.
I see it passed in here

https://github.com/ros2/system_tests/blob/5c2682ad721470a6d87fe178750061e3380f767d/test_communication/test/test_messages_c.cpp#L495-L496

But I don't see any calls to `test_msgs__msg__StaticArrayPrimitives__destroy`. Where is that being called?

yechun1, 2018-08-17:
@sloretz sorry that code scan tool we used is not free for sharing. The trace related with build code and is a little difficult to understand, below are copy-parse, would you please help to analyze and check if there is a potential issue.
Bullet 2:
* address_free: get_message frees address of expected_msg
https://github.com/ros2/system_tests/blob/5c2682ad721470a6d87fe178750061e3380f767d/test_communication/test/test_messages_c.cpp#L495-L496

* test_msgs__msg__StaticArrayPrimitives__init frees parameter msg
https://github.com/ros2/system_tests/blob/5c2682ad721470a6d87fe178750061e3380f767d/test_communication/test/test_messages_c.cpp#L443-L445

* builtin code
```
build/test_msgs/rosidl_generator_c/test_msgs/msg/static_array_primitives__functions.c
16 test_msgs__msg__StaticArrayPrimitives__init(test_msgs__msg__StaticArrayPrimitives * msg)
17{
// Condition !msg, taking false branch.
18  if (!msg) {
// Condition !rosidl_generator_c__String__init(&msg->string_values[i]), taking true branch.
36    if (!rosidl_generator_c__String__init(&msg->string_values[i])) {
// test_msgs__msg__StaticArrayPrimitives__destroy frees parameter msg 
37      test_msgs__msg__StaticArrayPrimitives__destroy(msg);
```
```
86 test_msgs__msg__StaticArrayPrimitives__destroy(test_msgs__msg__StaticArrayPrimitives * msg)
//  Condition msg, taking true branch.
88  if (msg) {
89    test_msgs__msg__StaticArrayPrimitives__fini(msg);
90  }
// free frees parameter msg
91  free(msg);
```

dirk-thomas, 2018-09-13:
The point @sloretz tried to make is that in the function [verify_message()](https://github.com/ros2/system_tests/blob/5c2682ad721470a6d87fe178750061e3380f767d/test_communication/test/test_messages_c.cpp#L493-L496) the variable is allocated on the stack and is only being initialized and populated in the function `get_message()` but nowhere is the destroy / fini function called (which would call `free`).

So based on the provided information it is still unclear where / what the problem is. So this needs more information and even better a reproducible example.

dirk-thomas, 2018-10-24:
@yechun1 Friendly ping.

yechun1, 2018-10-25:
@dirk-thomas, thanks for notice and sorry for update late.
just checked, the issue could not be reproduced with latest code base. so close it.

closed 2018-10-25 by yechun1