ros2/system_tests #291: Please add a README explaining how to add new tests

mkhansen-intel, 2018-08-15:
## Feature request
Add a README with some explanation of how to add new system tests

#### Feature description
I would like to be able to add a system test if I make an addition to rclcpp, but not completely sure how to do that. Please add basic instructions.

#### Implementation considerations
N/A

nuclearsandwich, 2018-08-30:
The system_tests repository is primarily meant to be internal-facing to the core team. If you're making changes to rclcpp or another repository which could benefit from additional system tests I would suggest raising the question of including additional system tests in the that pull request before opening a pull request here.

We are currently concentrating efforts on other aspects of ROS 2 development and the quasi-internal status of this repository sets it lower on the priority list of things which could use improved documentation.

If you are looking to implement additional system tests on your own you could start by using existing tests as a template.

For the rclcpp service tests [this block in CMakeLists.txt](https://github.com/ros2/system_tests/blob/33ef7fe5e5f17581ffff54a43fb017c83d8b4ca6/test_rclcpp/CMakeLists.txt#L281-L287) adds the test executables to the build and [this one](https://github.com/ros2/system_tests/blob/33ef7fe5e5f17581ffff54a43fb017c83d8b4ca6/test_rclcpp/CMakeLists.txt#L207-L213) adds them to the macro which invokes the test once per each rmw implementation being tested.

For this test the source for the [server executable](https://github.com/ros2/system_tests/blob/33ef7fe5e5f17581ffff54a43fb017c83d8b4ca6/test_rclcpp/test/test_services_server.cpp) is a normal ROS 2 node providing a service while the [client executable](https://github.com/ros2/system_tests/blob/33ef7fe5e5f17581ffff54a43fb017c83d8b4ca6/test_rclcpp/test/test_services_client.cpp) contains GTEST macros to perform tests on the service call results.

Since I've posted some information above, I'll try to reformat it to work as a minimal readme which recommends that contributors discuss new system tests in implementation pull requests before starting the pull request process here.

mkhansen-intel, 2018-08-30:
@nuclearsandwich - thank you, this answers my questions, you can close this issue if you like.

PR #294 (2018-08-30): Add a minimal readme linking to an example test.
Takes my comment from #291 and turns it into a very basic README.

connects to #291 
fixes #291
closed 2018-08-30 by nuclearsandwich