ros2/system_tests #319: FastRTPS/Connext Action tests are suspiciously short, and probably don't work

sloretz, 2018-12-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic
- Installation type:
  - source
- Version or commit hash:
  - https://gist.github.com/sloretz/6e5d7324b115da574d79c7e785d8968e
- DDS implementation:
  - Fast-RTPS + RTI Connext
- Client library (if applicable):
  - clcpp

#### Steps to reproduce issue

1. Run the action test using rclcpp with fastrtps and connext
```
colcon test --event-handlers console_direct+ --packages-select test_communication --ctest-args -R test_action_client_server__Fibonacci__rclcpp__rmw_fastrtps_cpp__rmw_connext_cpp
```

1. Run the executables the above test should be running on their own
   ```
   RMW_IMPLEMENTATION=rmw_fastrtps_cpp ./build/test_communication/test_action_client_cpp Fibonacci /foo/bar
   ```
   ```
   RMW_IMPLEMENTATION=rmw_connext_cpp ./build/test_communication/test_action_server_cpp Fibonacci /foo/bar
   ``` 

#### Expected behavior

1. The test would pass, taking about a second or so, and outputting info about goals and feedback
1. Run independently the tests should output information about goals and feedback

#### Actual behavior

1. The test passes in 0.1 seconds with no output
1. Run independently the client says it sent a goal, but the server does not print a message saying it received it.

#### Additional information

Same behavior if the rmw implementations are flipped so it's a connext client with fastrtps server. If both are fastrtps or both are connext it works as expected. The service/client and subscriber/publisher tests appear to work fine across fastrtps and connext.

sloretz, 2018-12-07:
It looks like all cross vendor tests are skipped if they involve services. If that's still the right thing to do, then it would make sense to skip action across vendors too since they use services.

https://github.com/ros2/system_tests/blob/9147079364b96d645f8d63df6f95cf3535ad7dad/test_communication/CMakeLists.txt#L191-L196

clalancette, 2020-11-18:
> It looks like all cross vendor tests are skipped if they involve services. If that's still the right thing to do, then it would make sense to skip action across vendors too since they use services.

Yeah, as far as I know, services don't currently work cross-vendor.  So I think disabling cross-vendor action testing is the correct thing to do here as well.