ros2/system_tests #359: Bad WStrings encoding when publishing over Connext on Windows

hidmic, 2019-05-06:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - From source
- Version or commit hash:
  - Current master at f65f852432090d1f2431d0cf496cbf4e2006922a
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclcpp & rclpy

#### Steps to reproduce issue
Run `rclcpp`-to-`rclpy` pub/sub tests using Connext on Windows:

```
colcon test --event-handlers=console_direct+ --packages-select test_communication --ctest-args " -R" " ^test_publisher_subscriber__rclcpp__rclpy__rmw_fastrtps_cpp"
```

#### Expected behavior
All tests pass.

#### Actual behavior
All test pass but it can be clearly seen that the subscriber fails to match incoming WString messages with the following traceback:

```
21: [test_subscriber-24]     raise RuntimeError('received unexpected message %r' % msg)
21: INFO:test_subscriber-24-stderr:[test_subscriber-24]     raise RuntimeError('received unexpected message %r' % msg)
21: --- Logging error ---
21: Traceback (most recent call last):
21:   File "C:\Python37\lib\logging\__init__.py", line 1037, in emit
21:     stream.write(msg + self.terminator)
21:   File "C:\Python37\lib\encodings\cp1252.py", line 19, in encode
21:     return codecs.charmap_encode(input,self.errors,encoding_table)[0]
21: UnicodeEncodeError: 'charmap' codec can't encode character '\u4e09' in position 136: character maps to <undefined>
21: Call stack:
21:   File "C:\Python37\lib\runpy.py", line 193, in _run_module_as_main
21:     "__main__", mod_spec)
21:   File "C:\Python37\lib\runpy.py", line 85, in _run_code
21:     exec(code, run_globals)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\launch_test.py", line 165, in <module>
21:     main()
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\launch_test.py", line 142, in main
21:     results = runner.run()
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\test_runner.py", line 231, in run
21:     results[run] = worker.run()
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\test_runner.py", line 143, in run
21:     self._launch_service.run()  # This will block until the test thread stops it
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\launch_service.py", line 350, in run
21:     self.__loop_from_run_thread.run_until_complete(run_loop_task)
21:   File "C:\Python37\lib\asyncio\base_events.py", line 571, in run_until_complete
21:     self.run_forever()
21:   File "C:\Python37\lib\asyncio\base_events.py", line 539, in run_forever
21:     self._run_once()
21:   File "C:\Python37\lib\asyncio\base_events.py", line 1775, in _run_once
21:     handle._run()
21:   File "C:\Python37\lib\asyncio\events.py", line 88, in _run
21:     self._context.run(self._callback, *self._args)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\launch_service.py", line 177, in _process_one_event
21:     await self.__process_event(next_event)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\launch_service.py", line 186, in __process_event
21:     entities = event_handler.handle(event, self.__context)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\event_handlers\on_process_io.py", line 75, in handle
21:     return self.__on_stderr(event)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\actions\execute_process.py", line 327, in __on_process_stderr
21:     self.__output_format.format(line=line, this=self)
21: Message: "[test_subscriber-24] RuntimeError: received unexpected message test_msgs.msg.WStrings(wstring_value='', array_of_wstrings=['1', 'two', '\u4e09'], bounded_sequence_of_wstrings=['one', '\u4e8c'], unbounded_sequence_of_wstrings=['.', '..', '...', '\u56db'])"
21: Arguments: ()
21: --- Logging error ---
21: Traceback (most recent call last):
21:   File "C:\Python37\lib\logging\__init__.py", line 1037, in emit
21:     stream.write(msg + self.terminator)
21:   File "C:\Python37\lib\encodings\cp1252.py", line 19, in encode
21:     return codecs.charmap_encode(input,self.errors,encoding_table)[0]
21: UnicodeEncodeError: 'charmap' codec can't encode character '\u4e09' in position 155: character maps to <undefined>
21: Call stack:
21:   File "C:\Python37\lib\runpy.py", line 193, in _run_module_as_main
21:     "__main__", mod_spec)
21:   File "C:\Python37\lib\runpy.py", line 85, in _run_code
21:     exec(code, run_globals)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\launch_test.py", line 165, in <module>
21:     main()
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\launch_test.py", line 142, in main
21:     results = runner.run()
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\test_runner.py", line 231, in run
21:     results[run] = worker.run()
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch_testing\Lib\site-packages\launch_testing\test_runner.py", line 143, in run
21:     self._launch_service.run()  # This will block until the test thread stops it
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\launch_service.py", line 350, in run
21:     self.__loop_from_run_thread.run_until_complete(run_loop_task)
21:   File "C:\Python37\lib\asyncio\base_events.py", line 571, in run_until_complete
21:     self.run_forever()
21:   File "C:\Python37\lib\asyncio\base_events.py", line 539, in run_forever
21:     self._run_once()
21:   File "C:\Python37\lib\asyncio\base_events.py", line 1775, in _run_once
21:     handle._run()
21:   File "C:\Python37\lib\asyncio\events.py", line 88, in _run
21:     self._context.run(self._callback, *self._args)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\launch_service.py", line 177, in _process_one_event
21:     await self.__process_event(next_event)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\launch_service.py", line 186, in __process_event
21:     entities = event_handler.handle(event, self.__context)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\event_handlers\on_process_io.py", line 75, in handle
21:     return self.__on_stderr(event)
21:   File "C:\Users\Administrator\nuclearsandwich_ws\install\launch\Lib\site-packages\launch\actions\execute_process.py", line 327, in __on_process_stderr
21:     self.__output_format.format(line=line, this=self)
21: Message: "[test_subscriber-24] RuntimeError: received unexpected message test_msgs.msg.WStrings(wstring_value='', array_of_wstrings=['1', 'two', '\u4e09'], bounded_sequence_of_wstrings=['one', '\u4e8c'], unbounded_sequence_of_wstrings=['.', '..', '...', '\u56db'])"
21: Arguments: ()
21: INFO:test_subscriber-24-stderr:[test_subscriber-24] RuntimeError: received unexpected message test_msgs.msg.WStrings(wstring_value='', array_of_wstrings=['1', 'two', '\u4e09'], bounded_sequence_of_wstrings=['one', '\u4e8c'], unbounded_sequence_of_wstrings=['.', '..', '...', '\u56db'])
21: [ERROR] [test_subscriber-24]: process has died [pid 5580, exit code 1, cmd 'C:\Python37\python.exe C:/Users/Administrator/nuclearsandwich_ws/src/ros2/system_tests/test_communication/test/subscriber_py.py WStrings /test_time_20_56_09'].
```

#### Additional information

This issue does not manifest itself on `rclcpp`-to-`rclcpp` nor `rclpy`-to-`rclpy` tests.

The issue is twofold, as not only an encoding/decoding step is wrong but the test shouldn't be passing. The latter is due to in an appropriately constructed assertion.

sloretz, 2019-05-06:
Are you already looking into this @hidmic ? I'm setting up a windows VM at the moment to try to reproduce it.

hidmic, 2019-05-06:
@sloretz I was planning to, but tomorrow. Feel free to take over.

closed 2019-05-09 by hidmic