ros2/system_tests #360: Failing rclpy-to-rclcpp WString pub/sub over Connext on Windows

hidmic, 2019-05-06:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - From source
- Version or commit hash:
  - Current master at f65f852432090d1f2431d0cf496cbf4e2006922a
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclcpp & rclpy

#### Steps to reproduce issue
Run `rclpy`-to-`rclcpp` pub/sub tests using Connext on Windows:

```
colcon test --event-handlers=console_direct+ --packages-select test_communication --ctest-args " -R" " ^test_publisher_subscriber__rclpy__rclcpp__rmw_connext_cpp"
```

#### Expected behavior
All tests pass.

#### Actual behavior

Subscription to a WString publisher fails with:

```
18: [test_subscriber-24] r
18: INFO:test_subscriber-24-stderr:[test_subscriber-24] r
18: [test_subscriber-24] eceived message does not match any expected message
18: INFO:test_subscriber-24-stderr:[test_subscriber-24] eceived message does not match any expected message
18: INFO:test_publisher-23-stdout:[test_publisher-23]
18: [ERROR] [test_subscriber-24]: process has died [pid 3156, exit code 3221226505, cmd 'C:/Users/Administrator/nuclearsandwich_ws/build/test_communication/Release/test_subscriber_cpp.exe WStrings /test_time_21_23_20'].
18: ERROR:test_subscriber-24:process has died [pid 3156, exit code 3221226505, cmd 'C:/Users/Administrator/nuclearsandwich_ws/build/test_communication/Release/test_subscriber_cpp.exe WStrings /test_time_21_23_20'].
```

#### Additional information

This issue does not manifest itself on `rclcpp`-to-`rclcpp` nor `rclpy`-to-`rclpy` nor `rclcpp`-to-`rclpy` tests.

The process terminates almost silently. Looking at the `rclcpp` subscriber [implementation](https://github.com/ros2/system_tests/blob/master/test_communication/test/test_subscriber.cpp#L51), this suggests that a crash occurs during `rclcpp::shutdown`.

hidmic, 2019-05-09:
Ok, I've came to the conclusion that this isn't actually related to `WString` support but to the fact that when the C++ subscriber test fails, it throws from within the callback right after `rclcpp::shutdown()`. That is, dropping [`throw std::runtime_error(...);`](https://github.com/ros2/system_tests/blob/935860e6d3bfc7f5aa8688272f3bf319560f0a44/test_communication/test/test_subscriber.cpp#L54) with a plain `return;` makes the stack corruption go away upon test failure.

hidmic, 2019-05-09:
So, the bad exit code comes from the abortion that results of the unhandled `std::runtime_error` exception. Adding a `try ... catch ...` clause is enough for the subscriber to finish gracefully. I'll PR it.

dirk-thomas, 2019-05-09:
Where is that exception thrown and why?

hidmic, 2019-05-09:
> Where is that exception thrown and why?

See [here](https://github.com/ros2/system_tests/blob/935860e6d3bfc7f5aa8688272f3bf319560f0a44/test_communication/test/test_subscriber.cpp#L54) (though I added the link in my comment above). As to why, it's the consequence of a message mismatch (which we had for the first time because of #359).

PR #363 (2019-05-09): Make test_subscriber_cpp always fail gracefully.
Fixes #360.
closed 2019-05-09 by hidmic