ros2/system_tests #407: building test_communication with other RMWs

adamsj-oci, 2020-03-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 68c495 (master)
- DDS implementation:
  - apparently only FAST-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

How does this work with `colcon`? I tried building with `RMW_IMPLEMENTATION` for something other than `rmw_fastrtps_cpp` but that doesn't seem to affect [this line](https://github.com/ros2/system_tests/blob/eloquent/test_communication/CMakeLists.txt#L73) which is how I assume I run `test_communication` against other RMWs.

----
## Feature request

#### Feature description

#### Implementation considerations

adamsj-oci, 2020-03-18:
Bad bug! I didn't need `$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` but I didn't forget to source the `local_setup.bash`. I had previously run `$ colcon list` from the command line and `$ colcon build ...` in a script which didn't have the same environment sourced. Once I fixed the environment, then I started to see output in the build logs.

tfoote, 2020-03-18:
Thanks for the follow up. Glad it's working.

closed 2020-03-18 by tfoote