ros2/system_tests #417: cppcheck timing out on test_rclcpp

nuclearsandwich, 2020-04-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 ARM64, macOS AMD64
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - n/a
- Client library (if applicable):
  - n/a

#### Steps to reproduce issue

Run test_rclcpp tests on ARM64 or macOS and you have a chance of the test timing out rather than completing. Here's an example build where I reached into ament_cppcheck and bumped the timeout up to 240. The test completed in 131s.

As seen on CI
* https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_debug/1174/testReport/(root)/test_rclcpp/cppcheck_xunit_missing_result/
* https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_extra_rmw_release/722/testReport/junit/(root)/test_rclcpp/cppcheck_xunit_missing_result/
* https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_release/1130/testReport/junit/(root)/test_rclcpp/cppcheck_xunit_missing_result/
* https://ci.ros2.org/view/nightly/job/nightly_osx_extra_rmw_release/732/testReport/junit/(root)/test_rclcpp/cppcheck_xunit_missing_result/

#### Expected behavior
cppcheck runs to completion

#### Actual behavior
cppcheck run times out before completion generating no test result output.
#### Additional information

chapulina, 2020-08-18:
> Here's an example build where I reached into ament_cppcheck and bumped the timeout up to 240.

I'm not sure what the "here" refers to, is there a link?

Is there a downside to increasing this timeout? Can a PR be opened with the change?

chapulina, 2020-08-19:
> Is there a downside to increasing this timeout? Can a PR be opened with the change?

Just found this: https://github.com/ament/ament_lint/pull/263