ros2/system_tests #432: [test_security] Fails compilation on second try

jacobperron, 2020-05-22:
## Bug report

**Required Info:**

- Operating System:
  -  Windows 10
- Installation type:
  - from source
- Version or commit hash:
  - b9bebaf21fd17f45bb29f3af9b619481d2d886b9
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Build test_security twice:
```
colcon build --merge-install --packages-select test_security
colcon build --merge-install --packages-select test_security
```

#### Expected behavior

Compiles twice without error.

#### Actual behavior

The second time, we see the following error:

```
--- stderr: test_security
CMake Error at CMakeLists.txt:341 (message):
  Failed to generate security artifacts: Traceback (most recent call last):

    File "C:\Users\jacob\dev\ws\ros\foxy\install\Scripts\ros2-script.py", line 11, in <module>
      load_entry_point('ros2cli==0.9.3', 'console_scripts', 'ros2')()
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\ros2cli\cli.py", line 67, in main
      rc = extension.main(parser=parser, args=args)
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\sros2\command\security.py", line 36, in main
      return extension.main(args=args)
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\sros2\verb\generate_artifacts.py", line 47, in main
      success = _artifact_generation.generate_artifacts(
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\sros2\api\_artifact_generation.py", line 31, in generate_artifacts
      if not _key.create_key(keystore_path, identity):
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\sros2\api\_key.py", line 86, in create_key
      _permission.create_permission_file(permissions_path, _utilities.domain_id(), policy_element)
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\sros2\api\_permission.py", line 65, in create_permission_file
      cert_content = _utilities.load_cert(cert_path)
    File "C:\Users\jacob\dev\ws\ros\foxy\install\Lib\site-packages\sros2\api\_utilities.py", line 131, in load_cert
      return x509.load_pem_x509_certificate(
    File "c:\python38\lib\site-packages\cryptography\x509\base.py", line 52, in load_pem_x509_certificate
      return backend.load_pem_x509_certificate(data)
    File "c:\python38\lib\site-packages\cryptography\hazmat\backends\openssl\backend.py", line 1222, in load_pem_x509_certificate
      raise ValueError(

  ValueError: Unable to load certificate.  See
  https://cryptography.io/en/latest/faq/#why-can-t-i-import-my-pem-file for
  more details.
```

#### Additional information

If I remove the build directory for test_security, then I can compile again.

dirk-thomas, 2020-06-04:
@ros-security-wg Can you please look into this and propose a fix. Thanks.

kyrofa, 2020-06-04:
Thanks for the ping. Need to find myself a Windows computer!

jacobperron, 2020-06-16:
I've noticed the same issue on macOS too.

mikaelarguedas, 2020-06-16:
See https://github.com/ros2/system_tests/pull/434 for a fix. @dirk-thomas and @jacobperron can you confirm if this fixes the issue for your use case ?

Arnatious, 2020-06-16:
Can't reproduce. Built twice and it was fine

mikaelarguedas, 2020-06-16:
> Can't reproduce. Built twice and it was fine

Can you try passing the `--cmake-force-configure` flag to your build ?
Similar to the command invocation from https://github.com/ros2/system_tests/pull/434#issuecomment-644997092

closed 2020-06-16 by jacobperron