ros2/system_tests #445: test_cli_remapping name_maker test failing

dirk-thomas, 2020-08-10:
See failing test in nightly: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1286/testReport/test_cli_remapping.TestCLIRemappingAfterShutdown/test_processes_finished_gracefully[_home_jenkins-agent_workspace_nightly_linux-aarch64_repeated_ws_src_ros2_system_tests_test_cli_remapping_test_name_maker/py_/

dirk-thomas, 2020-08-31:
@sloretz Any update on this?

sloretz, 2020-09-11:
Most recent failure https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1339/testReport/junit/test_cli_remapping.TestCLIRemappingAfterShutdown/test_processes_finished_gracefully[_home_jenkins-agent_workspace_nightly_linux-aarch64_repeated_ws_src_ros2_system_tests_test_cli_remapping_test_name_maker/py_/

I did not find any similar failures in the amd64 repeated job, so it's likely aarch64 specific

sloretz, 2020-09-11:
Seems to be [this test that is failing](https://github.com/ros2/system_tests/blob/2a0a6293ae186b5c1b1d75696f08bf472afb2930/test_cli_remapping/test/test_cli_remapping.py#L174-L176). It is failing because [`name_maker.py` is exiting with return code -2](https://github.com/ros2/system_tests/blob/master/test_cli_remapping/test/name_maker.py) ([SIGINT](https://man7.org/linux/man-pages/man7/signal.7.html)).

sloretz, 2020-09-11:
3 different tracebacks

```
03:23:43 1: [name_maker_node_name_replacement-7] Traceback (most recent call last):
03:23:43 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/system_tests/test_cli_remapping/test/name_maker.py", line 40, in <module>
03:23:43 1: [name_maker_node_name_replacement-7]     node = NameMaker()
03:23:43 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/system_tests/test_cli_remapping/test/name_maker.py", line 24, in __init__
03:23:43 1: [name_maker_node_name_replacement-7]     super().__init__('original_node_name', namespace='/original/namespace')
03:23:43 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 213, in __init__
03:23:43 1: [name_maker_node_name_replacement-7]     self._parameter_service = ParameterService(self)
03:23:43 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/parameter_service.py", line 40, in __init__
03:23:43 1: [name_maker_node_name_replacement-7]     node.create_service(
03:23:43 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 1302, in create_service
03:23:43 1: [name_maker_node_name_replacement-7]     service_capsule = _rclpy.rclpy_create_service(
03:23:43 1: [name_maker_node_name_replacement-7] KeyboardInterrupt
```

```
03:29:26 1: [name_maker_node_name_replacement-7] Traceback (most recent call last):
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/system_tests/test_cli_remapping/test/name_maker.py", line 40, in <module>
03:29:26 1: [name_maker_node_name_replacement-7]     node = NameMaker()
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/system_tests/test_cli_remapping/test/name_maker.py", line 24, in __init__
03:29:26 1: [name_maker_node_name_replacement-7]     super().__init__('original_node_name', namespace='/original/namespace')
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 210, in __init__
03:29:26 1: [name_maker_node_name_replacement-7]     self._parameter_service = ParameterService(self)
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/parameter_service.py", line 56, in __init__
03:29:26 1: [name_maker_node_name_replacement-7]     node.create_service(
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 1295, in create_service
03:29:26 1: [name_maker_node_name_replacement-7]     check_for_type_support(srv_type)
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
03:29:26 1: [name_maker_node_name_replacement-7]     msg_type.__class__.__import_type_support__()
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rcl_interfaces/lib/python3.8/site-packages/rcl_interfaces/srv/_set_parameters.py", line 284, in __import_type_support__
03:29:26 1: [name_maker_node_name_replacement-7]     module = import_type_support('rcl_interfaces')
03:29:26 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rosidl_generator_py/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 45, in import_type_support
03:29:26 1: [name_maker_node_name_replacement-7]     with add_dll_directories_from_env('PATH'):
03:29:26 1: [name_maker_node_name_replacement-7]   File "/usr/lib/python3.8/contextlib.py", line 240, in helper
03:29:26 1: [name_maker_node_name_replacement-7]     return _GeneratorContextManager(func, args, kwds)
03:29:26 1: [name_maker_node_name_replacement-7]   File "/usr/lib/python3.8/contextlib.py", line 83, in __init__
03:29:27 1: [name_maker_node_name_replacement-7]     self.gen = func(*args, **kwds)
03:29:27 1: [name_maker_node_name_replacement-7] KeyboardInterrup
```

```
03:28:11 1: [name_maker_node_name_replacement-7] Traceback (most recent call last):
03:28:11 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/system_tests/test_cli_remapping/test/name_maker.py", line 40, in <module>
03:28:11 1: [name_maker_node_name_replacement-7]     node = NameMaker()
03:28:11 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/system_tests/test_cli_remapping/test/name_maker.py", line 24, in __init__
03:28:11 1: [name_maker_node_name_replacement-7]     super().__init__('original_node_name', namespace='/original/namespace')
[... cut ...]
03:28:11 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/parameter_service.py", line 56, in __init__
03:28:11 1: [name_maker_node_name_replacement-7]     node.create_service(
03:28:11 1: [name_maker_node_name_replacement-7]   File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 1299, in create_service
03:28:11 1: [name_maker_node_name_replacement-7]     service_capsule = _rclpy.rclpy_create_service(
03:28:11 1: [name_maker_node_name_replacement-7] KeyboardInterrupt
```

jacobperron, 2020-12-03:
This failure hasn't happened since [September 6th](https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1339/testReport/junit/test_cli_remapping.TestCLIRemappingAfterShutdown/test_processes_finished_gracefully%5B_home_jenkins-agent_workspace_nightly_linux-aarch64_repeated_ws_src_ros2_system_tests_test_cli_remapping_test_name_maker/py_/). Closing for now, we can re-open if the failure reappears.

closed 2020-12-03 by jacobperron