ros2/system_tests #496: :man_farmer: `test_subscriber_terminates_in_a_finite_amount_of_time` and `test_processes_finished_gracefully` failing on Windows

Blast545, 2022-03-04:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - Source
- Version or commit hash:
  - main
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Check nightly windows jobs on the buildfarm.
Reference job showing an error in jenkins with both `test_subscriber_terminates_in_a_finite_amount_of_time` and `test_processes_finished_gracefully` failing.
https://ci.ros2.org/view/nightly/job/nightly_win_deb/2284/

#### Expected behavior
All test pass.

#### Actual behavior
These error appear constantly in the logs. It's flaky in the sense that the affected templated test differ from run to run, but most of the time at least one occur. However, important information, not all the times the error is shown in the jenkins report, for example, in the reference case attached above the following errors are shown in the UI:

```
projectroot.test_publisher_subscriber__rclpy__rmw_fastrtps_cpp
test_communication.TestPublisherSubscriber.test_subscriber_terminates_in_a_finite_amount_of_time[UnboundedSequences]
test_communication.TestPublisherSubscriberAfterShutdown.test_processes_finished_gracefully[UnboundedSequences]
test_communication.TestPublisherSubscriber.test_subscriber_terminates_in_a_finite_amount_of_time[WStrings]
test_communication.TestPublisherSubscriberAfterShutdown.test_processes_finished_gracefully[WStrings]
```

However, if you run grep on the log file looking for "FAIL: " you'll see the following list:

```
142: FAIL: test_subscriber_terminates_in_a_finite_amount_of_time[BoundedSequences] (test_communication.TestPublisherSubscriber)
142: FAIL: test_processes_finished_gracefully[BoundedSequences] (test_communication.TestPublisherSubscriberAfterShutdown)
143: FAIL: test_processes_finished_gracefully[Empty] (test_communication.TestRequesterReplierAfterShutdown)
142: FAIL: test_processes_finished_gracefully[Arrays] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_subscriber_terminates_in_a_finite_amount_of_time[Empty] (test_communication.TestPublisherSubscriber)
142: FAIL: test_processes_finished_gracefully[Empty] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_subscriber_terminates_in_a_finite_amount_of_time[Nested] (test_communication.TestPublisherSubscriber)
142: FAIL: test_processes_finished_gracefully[Nested] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_processes_finished_gracefully[Strings] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_processes_finished_gracefully[UnboundedSequences] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_subscriber_terminates_in_a_finite_amount_of_time[WStrings] (test_communication.TestPublisherSubscriber)
142: FAIL: test_processes_finished_gracefully[WStrings] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_subscriber_terminates_in_a_finite_amount_of_time[UnboundedSequences] (test_communication.TestPublisherSubscriber)
142: FAIL: test_processes_finished_gracefully[UnboundedSequences] (test_communication.TestPublisherSubscriberAfterShutdown)
142: FAIL: test_subscriber_terminates_in_a_finite_amount_of_time[WStrings] (test_communication.TestPublisherSubscriber)
142: FAIL: test_processes_finished_gracefully[WStrings] (test_communication.TestPublisherSubscriberAfterShutdown)
```

More info:
Templated tests: https://github.com/ros2/system_tests/blob/master/test_communication/test/test_publisher_subscriber.cpp
Actual test failures (problems related to finishing the process cleanly on windows): 
https://github.com/ros2/system_tests/blob/44ac714f2cd201b2a2c98949595e953d5a289191/test_communication/test/test_publisher_subscriber.py.in#L67
and  https://github.com/ros2/system_tests/blob/44ac714f2cd201b2a2c98949595e953d5a289191/test_communication/test/test_publisher_subscriber.py.in#L75

Crola1702, 2022-11-01:
Happened today in https://ci.ros2.org/view/nightly/job/nightly_win_deb/2532/

Crola1702, 2022-11-01:
Happened today in: https://ci.ros2.org/view/nightly/job/nightly_win_deb/2532/

Crola1702, 2022-11-07:
Today we had 6 of this here: https://ci.ros2.org/view/nightly/job/nightly_win_deb/2538/

Crola1702, 2022-11-10:
Today we had 3 of this here: https://ci.ros2.org/view/nightly/job/nightly_win_deb/2541/testReport/