ros2/system_tests #561: 🧑‍🌾 test_secure_publisher_subscriber fastrtps secure_comm timing out

Crola1702, 2025-03-04:
Reference build:
* https://ci.ros2.org/job/nightly_win_rep/3657/#showFailuresLink

Test regression:
* [test_security.test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0.xunit.missing_result](https://ci.ros2.org/job/nightly_win_rep/3657/testReport/junit/(root)/test_security/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0_xunit_missing_result/)

Log output:
```
test 42
      Start 42: test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0

42: Test command: C:\Python38\python.exe "-u" "C:/ci/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/ci/ws/build/test_security/test_results/test_security/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0.xunit.xml" "--package-name" "test_security" "--output-file" "C:/ci/ws/build/test_security/launch_test/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0.txt" "--env" "PYTHONDONTWRITEBYTECODE=1" "PATH="C:/ci/ws/install/Scripts" "C:/ci/ws/install/bin" "C:/ci/ws/install/opt/libyaml_vendor/lib" "C:/ci/ws/install/opt/libyaml_vendor/bin" "C:/ci/ws/install/opt/spdlog_vendor/lib" "C:/ci/ws/install/opt/spdlog_vendor/bin" "C:/Program Files/rti_connext_dds-6.0.1/bin" "C:/Program Files/rti_connext_dds-6.0.1/lib/x64Win64VS2017" "C:/ci/ws/install/opt/mimick_vendor/lib" "C:/ci/ws/install/opt/uncrustify_vendor/bin" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/HostX86/x64" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/HostX86/x86" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE/VC/VCPackages" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE/CommonExtensions/Microsoft/TestWindow" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE/CommonExtensions/Microsoft/TeamFoundation/Team Explorer" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/MSBuild/Current/bin/Roslyn" "C:/Program Files (x86)/Microsoft SDKs/Windows/v10.0A/bin/NETFX 4.8 Tools" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/Tools/devinit" "C:/Program Files (x86)/Windows Kits/10/bin/10.0.19041.0/x86" "C:/Program Files (x86)/Windows Kits/10/bin/x86" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools//MSBuild/Current/Bin" "C:/Windows/Microsoft.NET/Framework/v4.0.30319" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE" "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/Tools" "C:/xmllint/bin" "C:/Program Files/Git/cmd" "C:/Program Files/CMake/bin" "C:/Program Files/OpenSSL-Win64/bin" "C:/opencv/x64/vc16/bin" "C:/Windows/system32" "C:/Windows" "C:/Windows/System32/Wbem" "C:/Windows/System32/WindowsPowerShell/v1.0" "C:/Windows/System32/OpenSSH" "C:/cinc-project/cinc/bin" "C:/ProgramData/chocolatey/bin" "C:/opscode/chef-workstation/bin" "C:/Program Files/Cppcheck" "C:/ProgramData/chocolatey/lib/bullet/lib" "C:/ProgramData/chocolatey/lib/cunit/lib" "C:/ProgramData/chocolatey/lib/tinyxml2/lib" "C:/Python38/Scripts" "C:/Python38" "C:/Users/ContainerAdministrator/AppData/Local/Microsoft/WindowsApps"" "--append-env" "PATH=C:/ci/ws/build/test_security/Release" "--command" "C:/Python38/python.exe" "-m" "launch_testing.launch_test" "C:/ci/ws/build/test_security/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0_Release.py" "--junit-xml=C:/ci/ws/build/test_security/test_results/test_security/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0.xunit.xml" "--package-name=test_security"
42: Working Directory: C:/ci/ws/build/test_security
42: Test timeout computed to be: 20
42: -- run_test.py: extra environment variables:
42:  - PATH="C:/ci/ws/install/Scripts;C:/ci/ws/install/bin;C:/ci/ws/install/opt/libyaml_vendor/lib;C:/ci/ws/install/opt/libyaml_vendor/bin;C:/ci/ws/install/opt/spdlog_vendor/lib;C:/ci/ws/install/opt/spdlog_vendor/bin;C:/Program Files/rti_connext_dds-6.0.1/bin;C:/Program Files/rti_connext_dds-6.0.1/lib/x64Win64VS2017;C:/ci/ws/install/opt/mimick_vendor/lib;C:/ci/ws/install/opt/uncrustify_vendor/bin;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/HostX86/x64;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/HostX86/x86;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE/VC/VCPackages;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE/CommonExtensions/Microsoft/TestWindow;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE/CommonExtensions/Microsoft/TeamFoundation/Team Explorer;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/MSBuild/Current/bin/Roslyn;C:/Program Files (x86)/Microsoft SDKs/Windows/v10.0A/bin/NETFX 4.8 Tools;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/Tools/devinit;C:/Program Files (x86)/Windows Kits/10/bin/10.0.19041.0/x86;C:/Program Files (x86)/Windows Kits/10/bin/x86;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools//MSBuild/Current/Bin;C:/Windows/Microsoft.NET/Framework/v4.0.30319;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/IDE;C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/Common7/Tools;C:/xmllint/bin;C:/Program Files/Git/cmd;C:/Program Files/CMake/bin;C:/Program Files/OpenSSL-Win64/bin;C:/opencv/x64/vc16/bin;C:/Windows/system32;C:/Windows;C:/Windows/System32/Wbem;C:/Windows/System32/WindowsPowerShell/v1.0;C:/Windows/System32/OpenSSH;C:/cinc-project/cinc/bin;C:/ProgramData/chocolatey/bin;C:/opscode/chef-workstation/bin;C:/Program Files/Cppcheck;C:/ProgramData/chocolatey/lib/bullet/lib;C:/ProgramData/chocolatey/lib/cunit/lib;C:/ProgramData/chocolatey/lib/tinyxml2/lib;C:/Python38/Scripts;C:/Python38;C:/Users/ContainerAdministrator/AppData/Local/Microsoft/WindowsApps"
42:  - PYTHONDONTWRITEBYTECODE=1
42: -- run_test.py: extra environment variables to append:
42:  - PATH+=C:/ci/ws/build/test_security/Release
42: -- run_test.py: invoking following command in 'C:\ci\ws\build\test_security':
42:  - C:/Python38/python.exe -m launch_testing.launch_test C:/ci/ws/build/test_security/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0_Release.py --junit-xml=C:/ci/ws/build/test_security/test_results/test_security/test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0.xunit.xml --package-name=test_security
42: [INFO] [launch]: All log files can be found below C:\Users\ContainerAdministrator\.ros\log\2025-03-04-08-24-31-178657-13e24df09be8-8320
42: [INFO] [launch]: Default logging verbosity is set to INFO
42: test_subscriber_terminates_in_a_finite_amount_of_time (test_security.TestSecurePublisherSubscriber)
42: Test that the subscriber terminates after a finite amount of time. ... [INFO] [test_publisher-1]: process started with pid [7592]
42: [INFO] [test_subscriber-2]: process started with pid [10524]
42: [test_publisher-1] [INFO] [1741076671.248083300] [rcl]: Found security directory: C:\ci\ws\build\test_security\test\test_security_files\enclaves\publisher
42: [test_subscriber-2] [INFO] [1741076671.278028700] [rcl]: Found security directory: C:\ci\ws\build\test_security\test\test_security_files\enclaves\subscriber
42: [ERROR] [test_publisher-1]: process has died [pid 7592, exit code 3221225477, cmd 'C:/ci/ws/build/test_security/Release/test_secure_publisher_cpp.exe UnboundedSequences /'].
    Test #42: test_secure_publisher_subscriber__UnboundedSequences__rmw_fastrtps_cpp__secure_comm_0 ...............***Timeout  20.01 sec
```

fujitatomoya, 2025-03-04:
@Crola1702 is this happening only with windows?

Crola1702, 2025-03-05:
> @Crola1702 is this happening only with windows?

Yes!

alsora, 2025-03-13:
@Crola1702 we discussed this in the waffle meeting.
Let us know if you think we should bring up this in the ROS PMC meeting

closed 2026-10-19 by ahcorde